execuc/LCInterlocking #2: save_initial_objects fails when some of visible objects have no transparency attribute

arkhnchul, 2016-09-12:
when document have some objects which is "visible" in tree but aren't actually solids or so (constraints from Assembly2 as example), method save_initial_objects from treepanel.py fails with "object has no attribute 'Transparency'" stacktrace and workbench fails to activate. Preliminary check with " if hasattr(obj.ViewObject,"transparency"): " (or maybe trying/catching exception, i'm not pythonic at all) solves this.

execuc, 2016-09-12:
Thank you for finding this bug and for giving me the bug fix. I have implemented the first solution, with attributes checker. I have commited but I have not fully tested because I don't know how to create objects which are not solid.

arkhnchul, 2016-09-13:
for example, this file: it contains two solids and two objects from Assembly2 module (planeConstraint01 and planeConstraint01_), which have visibility but no transparency

[test1_assy2.zip](https://github.com/execuc/LCInterlocking/files/469880/test1_assy2.zip)

execuc, 2016-09-13:
I have opened the file but it does not show special objects because I don't have Assembly2 module (FreeCAD 0.14 on Debian Jessie).  If you test again this module with last commit, could you tell me if it ok ? Thanks a lot.

arkhnchul, 2016-09-14:
yeah, now it works as expected and ignore non-solid objects.

execuc, 2016-09-14:
Ok, thank you, I close the issue.

closed 2016-09-14 by execuc