execuc/LCInterlocking #3: Adding same parts fails due to thickness unequality

arkhnchul, 2016-09-14:
some parts fails to be added as "same parts" with message "_partname_ does not have the same thickness", even when thicknesses are equal in cad. This happens due to strong equality check between floats in append_link method of PartsList (partmat.py). Comparing absolute difference with some reasonable delta value, 0.001 or so, is more reliable here.
for example two parts in attached file, with thicknesses differ by 8.881784197e-16

[test1_assy2.zip](https://github.com/execuc/LCInterlocking/files/472084/test1_assy2.zip)

[thicknesses_unequality.patch.txt](https://github.com/execuc/LCInterlocking/files/472106/thicknesses_unequality.patch.txt)

execuc, 2016-09-14:
Ok, I fix it using a function defined in other file. Default tolerance is 10e-6 in this function but if issue is still present I will set to 0.001.

arkhnchul, 2016-09-29:
as far as i see, issue isn't showing anymore.

execuc, 2016-10-12:
Ok, thank you for the fix and the test.

closed 2016-10-12 by execuc