execuc/LCInterlocking #5: Fail adding hinges connection

arkhnchul, 2016-11-04:
Mod is unable to add hinges link to part which don't have opposite face equal to the one which is actually linking, because find_same_normal_face fails. It is supposed behaviour when building a linked chain, but sometimes last element differs from others - for example, chamfered box lid or so. It can be fixed by not finding last_face for very last part in the link chain - it is not used anyways.

Here attached example with such parts (rectangular and filleted one) and possible solution using counter.

[hingefail_test.fcstd.zip](https://github.com/execuc/LCInterlocking/files/572758/hingefail_test.fcstd.zip)
[hingefix_lastface.patch.txt](https://github.com/execuc/LCInterlocking/files/572757/hingefix_lastface.patch.txt)

execuc, 2016-11-05:
Ok I can reproduce it. It works if the hinge is from "roundedside" to "sqside" but it fails for reverse. I will study the problem and your solution. Thank you very much !

execuc, 2016-11-05:
Ok I commit the modification. There was also a problem to make hinge from "sqside" to "roundedsize" in do_intersection method. Cross product result was 10e-14 so it failed with zero equality test. I add a tolerance.
Could you tell me if hinges problem is fixed for you?

arkhnchul, 2016-11-07:
yeah, now it works well on part chains with complicated ending faces, like one that attached.
There is another issue with parts in this file - epsilon of 10e-6 in compare_value is too fine for arc recognition (diff_length_test variable in complete_hinges_properties function, makehinges.py). 10e-3 works fine.
[packer_hinge_example.fcstd.zip](https://github.com/execuc/LCInterlocking/files/574248/packer_hinge_example.fcstd.zip)

execuc, 2016-11-07:
Ok, it's now 10e-3 for this test. Thank you.

closed 2016-11-07 by execuc