execuc/LCInterlocking #6: assemble_list_element optimisation

arkhnchul, 2016-11-07:
it is possible to significantly increase overall performance by using Part.makeCompound method instead of fusing together all list elements one by one. It is especially important in hinges creation, where a big bunch of fusions occurs.
As i can see in my use-cases, it works same and not breaking anything.

[assemble_optim.patch.txt](https://github.com/execuc/LCInterlocking/files/576077/assemble_optim.patch.txt)

execuc, 2016-11-07:
Yes it's true, this module is not optimized and it may take long time.
I am just testing this modification, but in my case, I have a bad behavior when there were parts that intersecting themselves. 
I will do more tests but I think I will add this improvment as an option (checkbox), enable/disable by default.
What do you think of it ?

arkhnchul, 2016-11-08:
selectable option is good)
can you give an example of unwanted behaviour? maybe i can help a bit.

maybe it will be possible to create such separate function for using in hinge creation at least. I add some profiling output to make_hinges function to recognize main time-costly stages, and got such results (5 hinges per column, 20 links, 110 holes total, time in milliseconds):
x_positions 0
y_length 67.0
y_positions 0
y_positions_2 0
hinges_holes 7979
assembly 15522
transform 1

after two optimizations (mentioned here and one from issue #7 ):
x_positions 0
y_length 67.0
y_positions 0
y_positions_2 0
hinges_holes 156
assembly 0
transform 1

so needed time decreases around 100 times

execuc, 2016-11-08:
There are very good optimizations !

The problem occurs when T-Slot are created on opposite face of a same part. If screw length are too big, the two screw ways to remove on this part intersect themselve. Then it produces a generated part with only contours. I do not know if the explanation is understandable else I can make an image

However, I think that this new option (enable optimization) could be enabled by default in preview mode because it runs very fast. And maybe in slow mode when user has finish.

execuc, 2019-04-15:
It is now implemented for preview if "fast preview" checkbox is selected.

closed 2019-08-05 by execuc