execuc/LCInterlocking #14: Will you plan to make this available via FreeCAD addon manager ?

luzpaz, 2017-05-11:
https://github.com/FreeCAD/FreeCAD-addons

luzpaz, 2018-03-21:
CC @probonopd

execuc, 2018-11-28:
I do not think to integrate it into freecad addon manager because I'm not happy with the way I integrated it in the HMI.
Indeed I would have liked to see the parameters of this tool in the View part and not in another window. Which prevents to update the parameters of created parts without starting all over again.

luzpaz, 2018-11-28:
Would you be open to collaborate with the FreeCAD community, by posting your thoughts in a post and see if anyone would be interested in revising this ?

execuc, 2018-12-01:
I started to look at how to create items in the model tree view. I think it's not easy without too much modification. I will work on it.

execuc, 2018-12-02:
OK, thank you !

execuc, 2019-01-16:
Module is now improved. All tools can now be reedited.

luzpaz, 2019-07-26:
Available via FC AddonManager

closed 2019-07-26 by luzpaz