execuc/LCInterlocking #22: Great Extension - Please Improve md file

CBtiger, 2019-03-06:
I thinks what this workbench does is really useful. Its great to see that you are continuing to improve it. 

Please do not take offence, but I think the best way to improve it right now would to update the "Interocking" section in the desciption file. 

Suggestions 
* Rename "Interlocking" to "Slot", at least that is the name I can see in the tool tip
* Remove the video or say that it shows an older version with different layout and names. Today this makes the video totally confusing. 
* Where is the PreviewButton from the video? Its seems surprising that the newer version has not preview function any more. Of is there another way to do a preview?
* it is unclear what objects need to be added. The whole box? The 2 parts that should be joined? Does it work to put slots to only one part, too?
* Its totally unclear what face must be selected. I guess the face where the slots should be one. But the slots will affect 3 faces. Should one seclect all 3 faces?
 I think the most easy thing would be to give an example: Make a box and put 2 slots 10mm long, 5mm apart between a short side wall and the top of the box. For this example make clear what parts need to be added. Then what faces need to be added. Then what value must be set wheren. Any preview? 
When results are visible only after "OK", the next question a user would have. When I have made a mistake, how can I change it? Then again take the same example and give the steps to change the slots to 2 slots 8mm long and 8mm apart.
* Optionally I think its interesting to understand  what additional files are created. I am unclear if I could delete my original box files or not. Also I could not delete any of the new files with "Del", maybe its a bug or on purpose - I am not sure. 

With this you do not need a video or pictures but it will be an excellent and very helpful documentation. For every user with access to a laser cutter this will be an excellent workbench!

Thanks

execuc, 2019-03-08:
Hello,

Thank you for your interest and for your message to improve the module.
Indeed, I recently modified the module including HMI without changing the README and videos because of lack of time.

Here are some answers:
  - I removed preview feature to save time at the last development. But, it is now possible to validate a module and reedit it by clicking on the main object created in the treeview. So I told myself that there was no need for a preview.
  - The generator box is a convenient tool which allows to create a box without connection quickly but the user can use standard parts (from freecad Part workbenchs) and to assemble them with the interlocking tool.
  - All parts involved in connections (where a face is going to have a tab and where the piece will have a slot) must be defined.
  - I also found bugs to erase generated pieces. But I think that reopening the file worked.

I'm going to work on the README taking into account your remarks.

Thank you !

execuc, 2019-04-13:
Hello, 
I add a shortdocument for each module tool

luzpaz, 2019-05-27:
Some updates: 210b6250c899bd87ca4e90f9c1817a80a3cc0676

execuc, 2019-05-28:
> Some updates: [210b625](https://github.com/execuc/LCInterlocking/commit/210b6250c899bd87ca4e90f9c1817a80a3cc0676)

Thanks Luzpaz for the commit.

CBtiger, 2019-06-01:
The documentation is really excellent, now. 
Thanks a lot execuc for this great workbench

closed 2019-06-01 by execuc