execuc/LCInterlocking #27: This FC addon needs a legit svg icon for the FC AddonManager

luzpaz, 2019-07-26:
We bundle the FreeCAD-addons workbench/addons SVG icons within FC. Once you add it to this repo, I'll copy it in to master.

execuc, 2019-08-01:
Thanks for the integration of this module in freecad.
To understand what I need to do, I have to make an icon for this module in SVG to replace the 16x16 xpm icon ?

luzpaz, 2019-08-01:
not replace just have a svg version

execuc, 2019-08-05:
I added icon at base directory ("addon_icon.svg"). Could you tell me if that's okay ?

luzpaz, 2019-08-05:
It looks good but the size of the file is pretty large. Take for example other workbench icons in:  
https://github.com/FreeCAD/FreeCAD/tree/master/src/Mod/AddonManager/Resources/icons

execuc, 2019-08-05:
OK. I made a smaller one.

closed 2019-08-06 by luzpaz