execuc/LCInterlocking #29: Cannot remove splitter from empty shape

foertel, 2019-08-29:
>Running the Python command 'export_command' failed:
>Traceback (most recent call last):
>  File "/home/felix/snap/freecad/8/.FreeCAD/Mod/LCInterlocking/ExportPanel.py", line 78, in Activated
>    self.export_list(parts_list, new_doc)
>  File "/home/felix/snap/freecad/8/.FreeCAD/Mod/LCInterlocking/ExportPanel.py", line 94, in export_list
>    transform_shape(part, new_part, freecad_document)
>  File "/home/felix/snap/freecad/8/.FreeCAD/Mod/LCInterlocking/ExportPanel.py", line 49, in transform_shape
>    new_part_feature.Shape = part_feature.Shape.removeSplitter()
>
>Cannot remove splitter from empty shape

FreeCAD 0.18 (16093) with LCInterlocking from the addon manager.

It happens, when I try to export all objects. If I just choose the active Bodies, it works okay. As soon as your drawing gets a little more complex (like a lot of cutouts and stuff), it's hard to export all the right objects. Maybe we can add a filter to ignore empty shapes?

luzpaz, 2019-11-26:
@foertel can you test with FreeCAD v0.19 and report ?

luzpaz, 2020-03-01:
@foertel please follow-up

undercoverdesigns, 2020-10-02:
I can verify that this happens in 0.19, as well.  If I select the Multijoin, I get this error.  If I select the individual *_tab objects, export works fine.