execuc/LCInterlocking #30: Maximum recursion depth exceeded in comparison

rakotoxen, 2019-09-25:
Starting from a blank project in Freecad 0.19 daily, navigating to the Laser Cut Interlocking workbench, and clicking the "Make box without tab" button, I get the following in the Report View:

`Traceback (most recent call last):
  File "/usr/share/freecad-daily/Mod/LCInterlocking/MakeBoxPanel.py", line 51, in onChanged
    if prop == "recompute":
<class 'RecursionError'>: maximum recursion depth exceeded in comparison
Running the Python command 'make_box_command' failed:
Traceback (most recent call last):
  File "/usr/share/freecad-daily/Mod/LCInterlocking/MakeBoxPanel.py", line 192, in Activated
    vp.setEdit(ViewProviderGroupBox)
  File "/usr/share/freecad-daily/Mod/LCInterlocking/MakeBoxPanel.py", line 85, in setEdit
    FreeCADGui.Control.showDialog(MakeBox(self.Object))
  File "/usr/share/freecad-daily/Mod/LCInterlocking/MakeBoxPanel.py", line 153, in __init__
    self.preview()
  File "/usr/share/freecad-daily/Mod/LCInterlocking/MakeBoxPanel.py", line 172, in preview
    self.obj_box.recompute = True

maximum recursion depth exceeded in comparison`

Python console displays the following:

`Python 3.6.8 (default, Aug 20 2019, 17:12:48) 
[GCC 8.3.0] on linux
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> Gui.runCommand('Std_ViewStatusBar',1)
>>> Gui.runCommand('Std_ViewStatusBar',0)
>>> Gui.runCommand('Std_Workbench',18)
>>> ### Begin command Std_New
>>> App.newDocument("Unnamed")
>>> App.setActiveDocument("Unnamed")
>>> App.ActiveDocument=App.getDocument("Unnamed")
>>> Gui.ActiveDocument=Gui.getDocument("Unnamed")
>>> Gui.activeDocument().activeView().viewDefaultOrientation()
>>> ### End command Std_New
>>> Gui.runCommand('Std_OrthographicCamera',1)
>>> ### Begin command Std_Workbench
>>> Gui.activateWorkbench("LCInterlockingWorkbench")
>>> ### End command Std_Workbench
>>> Gui.runCommand('make_box_command',0)
>>> Gui.runCommand('Std_About',0)
>>> `

My version and OS info here:

`OS: Ubuntu 18.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.
Build type: Release
Python version: 3.6.8
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)`

rakotoxen, 2019-09-25:
Same thing occurs if I try "Make rounded box without tab"

execuc, 2019-09-30:
Ok, sorry for the issue. I would test when I could.

luzpaz, 2019-11-26:
@execuc can you recreate this?

execuc, 2019-11-26:
Yes I have the problem in 0.19, I will investigate. Sorry for the problem.

execuc, 2019-11-28:
Ok I have corrected the problem in the last commit.

closed 2019-11-28 by execuc

luzpaz, 2019-11-29:
> Ok I have corrected the problem in the last commit.

What commit is the fix in?

execuc, 2019-11-29:
Sorry, I had a problem in the last commit. I just commited it.

luzpaz, 2019-11-29:
https://github.com/execuc/LCInterlocking/commit/8d067f891c6a8e4244a2129915e2c395f70cd017