execuc/LCInterlocking #31: [bug] SVG export converts circle to square

Foadsf, 2019-09-30:
I'm not sure if this is a bug in this workbench or is there any issues with the SVG export. but when I export the drawing of [this part](https://forum.freecadweb.org/download/file.php?id=92902) as SVG the circle is converted to a square!

[image]

[here the SVG file](https://forum.freecadweb.org/download/file.php?id=92903)

[image]

I have explained the issue with more details [here in the FreeCAD forum](https://forum.freecadweb.org/viewtopic.php?f=3&t=39719&p=337613#p337613).

execuc, 2019-09-30:
Hello, in May 2019, I had the same problem since FreeCAD 0.18. I have not tried since. I did not have a problem with 0.17
I marked a note in the export.md help file:

>  Some elements of projected parts have problems. For example, a circle may be displayed and exported in flatten SVG as hexagon. A workaround is to export in Autodesk DXF 2D (.dxf) format. It may be imported as inkscape but elements can not be ungrouped. To do it, dxf file has to be opened in librecad, then select all (Ctrl+a) and click on Tools -> Modify -> explode. Save file and re-import with inkscape. Now each segments and curves can be edited.

I have see the post in freecad forum it is maybe a bug in freecad.

luzpaz, 2019-11-26:
This is an upstream (freecad) bug that is fixed in FreeCAD v0.19, so this ticket can be closed.  
Perhaps mention on the README for users to use the latest v0.19 with this Addon (if this addon is compatible with it?).

execuc, 2019-11-26:
@luzpaz I will update README for 0.19 as soon as I found the solution to the ticket #30 which blocks the use of this addon for this version.

luzpaz, 2020-03-01:
Looks like #30 is closed. Shall we close this ticket?

execuc, 2020-04-29:
Tested with FreeCAD_0.19-20802-Linux-Conda_glibc2.12-x86_64.AppImage, SVG export works.

closed 2020-04-29 by execuc

rablack, 2021-02-16:
There seems to be a workaround for this issue in 0.18 without resorting to dxf.

1. In Preferences->Import-Export go to the SVG tab
2. Edit the setting for "Max segment length for discretized arcs" to be something sensible greater than 0.00

As far as I can tell arcs that FreeCAD 0.18 doesn't know how to render as SVG are replaced with a series of straight-line segments. If you have the value of the setting mentioned above set to 0.00 FreeCAD appears to try to guess what segment length you want and it sometimes seems to guess wrongly.

I don't know whether this fixes every instance, but I had more than 10 problem models, all of which were fixed by setting the value to 0.1mm.

reox, 2023-04-07:
> I don't know whether this fixes every instance, but I had more than 10 problem models, all of which were fixed by setting the value to 0.1mm.

I'm in 0.20 and had to use 0.1mm as well, otherwise all circles were exported as polygons...
Maybe it would be a good idea to add this in the documentation of the module?