execuc/LCInterlocking #32: please add donation information

Foadsf, 2019-10-01:
I'm trying to collect all the FLOSS CAD developers who have Patreon, Flattr, Liberapay, PayPal... [here ](https://www.reddit.com/r/opensource/comments/ckkl5e/supporting_free_libre_and_open_source_cad/?utm_source=share&utm_medium=web2x) in this post so people can support them. I would appreciate if you could also create an account for any of the services above or share other method people can support you developing this workbench.

iteles, 2020-10-06:
We're just going to start trying out this workbench now and would also love to support the development 😊

Tectu, 2020-10-06:
Same here! This workbench finally allows me to use FreeCAD for lasercutting projects. I'd also throw a couple of bucks in the hat.