execuc/LCInterlocking #37: How to update the thickness after multijoin

nickoe, 2019-12-08:
Hello, thank you for making such a great tool, it works really nicely.

I made a box design, but I found that I choose the wrong material thickness. Is there a way to change this?

I tried to change it on the Box object and it do seem like the thickness is correctly reflecting in those objects and the Multijoin_origin_parts as well, but the objects with the tabs specified before I changed the thickness are not updated.

Is there a way to propagate the thickness changes to those objects?

I am using f1bd7f7c8831b4ee13462a00e2c29c1e7126bb65

execuc, 2019-12-09:
Hello,
Thank you very much !

Is the box created with the tool of this module "make box without tab"? If yes, I managed to change the thickness from the "box" object, and then to change the thickness of parts involved in the "MultiJoin" object.

Could you specify the version of freecad? Do you have a freecad file to test the problem or a procedure to follow ?

nickoe, 2019-12-09:
I am not home now, but I used the appimage version of freecad downloaded last week or so, meaning something like freecad version 0.18.

My model is available at https://github.com/nickoe/misc_3d_things/blob/6b1313ef2b1f338ba0dbd984286bd9a8fca32819/rack/1U_lasercut_box.FCStd

nickoe, 2019-12-09:
@execuc The freecad version is:
OS: Arch Linux
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16146 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.18.4)
Hash: 980bf9060e28555fecd9e3462f68ca74007b70f8
Python version: 3.6.7
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

nickoe, 2019-12-09:
Here is a picture trying to illustrate the issue a bit more clearly. You can see there is a gab for the selected part to the rest. It should be thicker such that it would touch the other parts here.
[image]

nickoe, 2019-12-16:
@execuc I am not trying to be too pushy, but do you see what I mean?

execuc, 2019-12-17:
Yes I see what you mean but I have verry little time currently. I am sorry.

If you double-click on Multijoin object, then change material thickness of each part and click OK to recompute all, it doesn't work ?

luzpaz, 2020-03-01:
@nickoe ping