execuc/LCInterlocking #38: Interlocking doesn't take PartDesign Placement into account

oliboss, 2019-12-31:
When creating parts in the Part Design Workbench they are usually padded from a Sketch. 
In order to place them in the right position for the joint. The Placement in PartDesign is based on the Placement of the Body-Container, not the Pad itself. 

The interlocking workbench uses the placement of the Pad inside of the Body (which is 0) -- so that the joint can't be created. 

[multijointest.zip](https://github.com/execuc/LCInterlocking/files/4012809/multijointest.zip)

execuc, 2020-01-28:
Hello,

Thanks for the example file.
I try to reproduce the problem but it works when I create an interconnection between Pad001_tab and Pad_tab. I'm not sure I understood the problem.

Could you explain the procedure to have the problem with this file, please ?

Thanks

luzpaz, 2020-03-01:
@oliboss please follow up

mac-call, 2020-05-08:
Hello, 
I  think i have the same issue and didn't know what "interconnection" mean here. (I'm quite new to freecad).
If you have two "bodies" which are not in the right place to "interlocked" you can move them using the "placement -> position" of one body : 
[image]

With the correct X, Y  and Z in "position", Body001 is in the right place : 
[image]

You can add the two "bodies" as "same parts" in LCINterlocking but then, you can't add any "faces". Don't know why but I suppose there is a good reason.

If you add the two parts of the bodies ("Pad" and "Pad001") as "same parts", you can add the correct "face" to LCInterlocking, but "Pad001" loose the offset position of its parent "Body001".
And you end up with something like that (with Pad001_tab having no offset position): 
[image]

Using the "make box without tab" tool that don't generate "bodies", and with all parts well positioned relatively to each other,  is  OK.

Probably missing something big here, but I'm quite stuck for many hours on this.
Great tool nevertheless :)
Thanks in advance.

shermelin, 2020-06-22:
Hello,

first, thanks for this workbench which saves quite some time!!!

I have the same issue here I think: several bodies in a part container, when trying to make the interlocking tabs, the created parts move to the origin of the part (or some other origin).

Here is my test file; trying to interlock "fond" with any of the other "etage0x")
[test-LCI.zip](https://github.com/execuc/LCInterlocking/files/4813360/test-LCI.zip)

execuc, 2020-06-22:
Hi,
Thank you for your interest in this workbench.
OK, I will watch as soon as I can. Thanks for the example attached.

acidspit, 2020-06-28:
Hi,
Thanks for creating this module. I think my problem may be the same or related to the ones stated above. When creating tabs on the lower layer of the attached file everything works fine, but for all parts in the upper layer no tabs are created. Am I missing something? Thanks for your help.

Version
OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.4 (GitTag)
Build type: Release
Branch: releases/FreeCAD-0-18
Hash: 980bf9060e28555fecd9e3462f68ca74007b70f8
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: German/Germany (de_DE)

[test.zip](https://github.com/execuc/LCInterlocking/files/4842227/test.zip)

shermelin, 2020-07-02:
Thanks @execuc.

I've tried to have a look, there's a method for Part::Features IIRC named getGlobalPlacement() which appears to give the relevant info. However, I had troubles to follow all the details in the module and could not yet find where to use it...

execuc, 2020-07-04:
Hello ! Sorry for the wait.

For @shermelin, effectively it is a problem of global placement. Transformation matrix of root container does not seem to be taken into account. I'm sorry for this problem because I never used a root container as you. A quick workaround is to create a simple copy of parts from Part workbench / Part tab / Create a simple copy and to make interlocking from these parts.
There is a pending pull request which may correct the problem and I have to integrate. I have to get back into the code !

For @acidspit, I think the problem is different because there are "simple parts" (== not attached to root container with transformation). In Interlocking module, when I am trying to add some of your top parts with "Add same parts" then I have a log which indicates parts don't have the same thickness. In fact, the module incorrectly computes the thickness of the part T (= 63.07 instead of 3mm) and I don't know why ! I have to debug. If I copy part B and use it instead of part T, it works better. For information, the module may find it difficult to manage the parts with rounding and with non-necessary faces like the one circled in red in the image. 

[image]

acidspit, 2020-07-06:
> Hello ! Sorry for the wait.
>
> For @shermelin, effectively it is a problem of global placement. Transformation matrix of root container does not seem to be taken into account. I'm sorry for this [...]

Hi thanks for the reply. With your hint regarding the simple parts and the edges, I got it working. Concerning the thickness of part "T" I simply recreated it.

shermelin, 2020-07-12:
Thanks @acidspit and sorry for the late reply.

That seems to do the trick!