execuc/LCInterlocking #39: Interlocking is not working in freecad 0.18.4

fabricioncosta, 2020-01-27:
Hi, 

I could not use this functionality. I follow all the steps described  on the tutorial (https://www.youtube.com/watch?v=YGFIdLpdWXE&feature=youtu.be)  and it did not generating any error message, but the final result is the same box I have created (without interlocking)

https://www.youtube.com/watch?v=YGFIdLpdWXE&feature=youtu.be
Freecad: 0.18.4.

Could you help me? 

Thanks,

execuc, 2020-01-28:
Hi,

The video is old and does not show the change I made in the interface. Now, a 'Multijoin' object is created at the end of the interlocking tool.

In interlocking tool, do you have created the "faces" (with 'add faces' or 'add same face' button) in the main window and not in the preview window (preivew button) ?

Do not hesitate to upload print screens or FreeCAD file which could help me reproduce the problem.

Thanks

fabricioncosta, 2020-01-28:
Hi,
Multijoin object continue to be created at the begining (before selected the parts in interlocking tool). I am using freecad 0.18.4 and the last version of LCIterlocking installed manually.  I have just created a box (make a box without tab) with this parameters:

[image]

and runned "interloking" with all the parts added:

[image]

The final result do not have the tabs and pads, as expected:

[image]

Thanks,

execuc, 2020-01-29:
Hi,

Thank you for describing the problem with images which I understand.
In fact, "interlocking" tools has two steps:
- Define all the parts and set options (thickness, laser beam...)
- Declare all the faces which must have tabs.

I think it's this second step that is missing. For each faces (or group of faces), you have to select them and click on 'add face' (or 'add same face') . Three types of tab exist and must be selected from the drop-down menu before pressing the 'add face' button. Depending on the type of tab, you must configure the tab options (size, repetition ...

Sometimes it's a bit tricky to select faces because you have to hide parts with the space key and then display them again

There is a short documentation for this tool: https://github.com/execuc/LCInterlocking/blob/master/docs/interlocking.md

fabricioncosta, 2020-01-29:
Hi,

Now I could generate the interlocking! 

[image]

Thank you very much!

execuc, 2020-01-29:
You're welcome !

closed 2020-01-29 by execuc