execuc/LCInterlocking #43: Crosspiece shall allow objects of different height

euphi, 2020-10-04:
Is it not possible to create a crosspiece of two parts of different height, because this raises a "not managed" exception:

```
Traceback (most recent call last):
  File "/home/ian/.FreeCAD/Mod/LCInterlocking/panel/crosspiece.py", line 54, in onChanged
    self.execute(fp)
  File "/home/ian/.FreeCAD/Mod/LCInterlocking/panel/crosspiece.py", line 136, in execute
    computed_parts = make_cross_parts(parts)
  File "/home/ian/.FreeCAD/Mod/LCInterlocking/lasercut/crosspart.py", line 372, in make_cross_parts
    else:
<class 'ValueError'>: Not managed
```

However, it is easily possible to create such a crosspiece. I just added the call to `remove_intersections(first_part, second_part, referential_faces, axis)` also for the "not managed" case and the result is fine (with the first part higher than the second one)

[image]

Obviously this is not the clean solution, but I don't really understand the different cases in `make_cross_parts(parts)`, so I can't create a better solution.

execuc, 2020-10-07:
Hello, I think I put this case aside and never came back to it ! Thank you for this use case and your solution. I will watch as soon as possible.

jvanbruegge, 2023-09-18:
Just hit this as well and replacing the `Not manged` error with `remove_intersections(first_part, second_part, referential_faces, axis)` in `lasercut/crosspart.py` worked for me as well