execuc/LCInterlocking #44: Slot width tolerance wrong way around

Tectu, 2020-10-06:
I'm experiencing issues with the `Slot width tolerance` setting of the tab feature.

It seems like the the `slot width tolerance` is applied "the wrong way around". Instead of the slot being wider than the tab (or the tab shorter than the slot) I end up with tabs that are too wide. After applying the operation the tabs are made wider by the amount specified in the `slot width tolerance` parameter.

I imagined that specifying a negative slot width tolerance would be a sufficient workaround but the field only accepts positive numbers.

I'm experiencing this issue with FreeCAD 0.18.4 and 0.19.22.

Am I missing something / doing something wrong?

execuc, 2020-10-07:
Hello, I did a test and it seems to work as you specify it. When I increase the "Slot width tolerance" parameter of the part, the slot width increases while the tab remains the same. Attached is a screenshot with a high value (5mm).

[image]

There is the opposite behavior when I increase the "laser beam diameter" parameter, wouldn't there be a shift in the parameters (from the interface) ?

Tectu, 2021-01-03:
I can confirm that this is working as intended. The issue I described arises when the `slot width tolerance` is smaller than the `laser beam diameter` value.

closed 2021-01-03 by Tectu