execuc/LCInterlocking #54: Does not work

MikePryadko, 2021-09-18:
### System
OS: Ubuntu 18.04.6 LTS (XFCE/xubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.
Build type: Release
Branch: unknown
Hash: 7a9aa7edf2f44e3eb6a8e3e61c753368496b1399
Python version: 3.6.9
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.5.2
Locale: English/UnitedStates (en_US)

### Problem
I'm using [this video](https://youtu.be/YGFIdLpdWXE) as tutorial, but on my machine nothing happens when I click "Add same faces" :neutral_face: 

### Demonstration

https://user-images.githubusercontent.com/31248040/133878080-4816585d-5c20-4557-9ca5-ffec3223725c.mp4

shahabaz, 2021-11-02:
I am not an expert here, Just a suggestion Try a lower version of Freecad see if it works. I am also facing a different problem. I am suspecting this is version compatibility issue.

martonmiklos, 2022-02-01:
> ### System
>
> OS: Ubuntu 18.04.6 LTS (XFCE/xubuntu) Word size of OS: 64-bit Word size of FreeCAD: 64-bit Version: 0.19. Build type: Release Branch: unknown Hash: [...]

Check the report pane.
With 0.19 I got 
``` Part of face Face1 is not configured ``` messages

RBeut, 2022-07-25:
I had the similar issue.

What helped in my case was making sure that i selected the face of a geometry and not the whole geometry from the combinational view

example of full body selection
[image]

example of only face selection

click in the model on the face you would want to select

[image]

NicoPy, 2023-03-03:
You do not use the tool correctly :
- Select both parts.
- Click "Add parts" or "Add same parts".
- Hide the left part (select it and hit "space" key)
- Select the face of the right part in contact with the left part.
- Click "Add faces" or "Add same faces".
- Select the added face in the tree
- Trim parameters (Number of tabs, width of tabs...)
- Click "Preview"
- Adjust parameters if needed
- Click "OK" when done