execuc/LCInterlocking #61: Export problem : some parts are not correctly oriented

NicoPy, 2023-03-03:
Hi,

Some parts of my quite simple design are not exported correctly (with the correct orientation).

Here is my design :
[image]

Here is the output of the "Export" command :
[image]

Why are the 2 tubes not exported with the same orientation than the other part ?

NicoPy, 2023-03-03:
Here is my design :
[Molette.zip](https://github.com/execuc/LCInterlocking/files/10886836/Molette.zip)

NicoPy, 2023-03-04:
I've found the answer to my question but no solution.

The answer is in the source code : The extension uses the face with the biggest area as a reference. In my case, the external face of the tube is the biggest one. 
Changing the inner radius of a tube solves the problem.
In the following picture, face A is bigger than face B but smaller than face C.
[image]

This does not solve my problem since changing the inner radius of the tubes is not what I want. 
The only solution is to make tubes thinner just for the export (and revert once done). This trick is manageable with this very simple designs but can't be applied with many other designs.

Any idea for a better algorithm (not using biggest face as reference) ?

s-light, 2023-08-18:
> Any idea for a better algorithm (not using largest face as reference) ?

use a user-selected face - maybe auto-suggest one...
all other magic automation will not work in some scenarios...

NicoPy, 2023-08-18:
Yes, I think you are right.